SJSURoboticsTeam/urc-control-system-2023 #6: Dosing pump driver

Coreyboy1820, 2023-01-16:
The science bot has an injector for the different chemicals, so we need a driver, heres a link but it may change: https://www.amazon.com/gp/product/B01IUVHB8E/ref=ox_sc_act_title_2?smid=A34K5WF5Z9R33P&psc=1

Coreyboy1820, 2023-01-28:
Dosing pump is finalized, the link is to the correct version.

PhazonicRidley, 2023-01-29:
Awaiting electrical's motor controller board and libhal's lpc4078 pwm API implementation.

See: [This Commit](https://github.com/SJSURoboticsTeam/urc-control-system-2023/commit/243194ca43031942f345c71951b8bdd62b41ad3e)

PhazonicRidley, 2023-02-01:
Need to certain specific information about controller board. Need PWM duty cycle, max frequency, frequency range

william-green, 2023-02-18:
This appears to be the datasheet for this pump.
[ROB-10398_eng_tds.pdf](https://github.com/SJSURoboticsTeam/urc-control-system-2023/files/10775197/ROB-10398_eng_tds.pdf)

william-green, 2023-02-18:
Overview of the science robot: samples and pump mechanisms.
[image]

william-green, 2023-02-19:
Running low on PCA. Will order.

william-green, 2023-02-19:
Here is an overview of the pump system. For development/prototyping, we're using a PCA board but, in the final product, the PCA chip is integrated into the science board.
[image]

closed 2023-02-27 by william-green