SJSURoboticsTeam/urc-control-system-2023 #7: Air pump driver

Coreyboy1820, 2023-01-16:
The science team needs an air pump and it needs a driver, heres the amazon link for it: https://www.amazon.com/gp/product/B01MT7B2A8/ref=ox_sc_act_title_1?smid=A1V3K7OI49CJC5&psc=1

Coreyboy1820, 2023-01-28:
Air pump is finalized, the link is to the correct version.

PhazonicRidley, 2023-01-29:
Awaiting electrical's motor controller board and libhal's lpc4078 pwm API implementation.

See: [This Commit](https://github.com/SJSURoboticsTeam/urc-control-system-2023/commit/243194ca43031942f345c71951b8bdd62b41ad3e)

PhazonicRidley, 2023-02-01:
Need to certain specific information about controller board. Need PWM duty cycle, max frequency, frequency range

Pinyiyeh, 2023-02-23:
The PWM frequency for the air pump and dosing pump should be 32kHz.

kammce, 2023-02-23:
The maximum frequency of the PCA PWM generators is 1.56kHz, so 32kHz is not achievable.

Pinyiyeh, 2023-02-23:
My mistake. We state the frequency required to drive the pump motor on the schematic. 
The air pump and dosing pump motor should be driven at 1.5kHz.

kammce, 2023-02-24:
No no, not your mistake, just unfortunate that we cannot achieve higher voltages.

Pinyiyeh, 2023-02-24:
Yeah, 1.5kHz is too low for a DC motor.

kammce, 2023-02-24:
Yeah thats why I brought it up during the board review. You can run them at 1.5kHz, but when you are stepping them down at that frequency it can be very problematic.

closed 2023-02-27 by william-green