SJSURoboticsTeam/urc-control-system-2023 #8: Hall effect driver

Coreyboy1820, 2023-01-16:
The science team needs a driver for the hall effect, research this and create a driver for it. https://www.amazon.com/gp/product/B07QS6PN3B/ref=ox_sc_act_title_1?smid=A34K5WF5Z9R33P&psc=1

Coreyboy1820, 2023-01-28:
This is really just creating a demo to see how it works and if it even does work.

Coreyboy1820, 2023-01-28:
Hall effect sensor is finalized, this is the correct sensor.

william-green, 2023-02-12:
@Coreyboy1820 I made a pull request for a working demo. Let me know what the next steps are. Should I make a wrapper class for abstraction purposes? Thanks

Here is the circuit schematic used.
[image]

william-green, 2023-02-18:
Finished.

closed 2023-02-18 by william-green