SJSURoboticsTeam/urc-control-system-2023 #72: Inverse kinematics for controlling arm from joystick-serial

starphys, 2023-03-18:
- [ ] Basic prototype - renders in MC, can choose points in xyz, calculates angles for arm servos
- [ ] Controlling arm - write calculated angles to server, avoid impossible states
- [ ] UI - point selection feels natural
- [ ] Feedback - update angles from rover
- [ ] 3D Model (stretch goal) - arm displays on top of rover in MC, accurately enough to avoid collisions with legs

starphys, 2023-03-18:
@Viha123 I've got some of this working (barely) in the [inverse-kinematics](https://github.com/SJSURoboticsTeam/joystick-serial-prototype-2022/tree/inverse-kinematics) branch. The next things we need to figure out are how to properly construct our chain, how to put the correct constraints on it, and how to convert the output from [result](https://github.com/SJSURoboticsTeam/joystick-serial-prototype-2022/blob/81d8a1ba599275615e98719ddf3533d08fddfcee/src/components/InverseKinematics.tsx#L53) to our format for arm angles.

starphys, 2023-03-20:
The control system (in Target.jsx) needs to be updated to only accept input within the correct component, and to scale the x/y coordinates proportionally to that component instead of to the whole window. Right now clicking on the center of the screen is the only way to send input at the center of the model, even when it is not actually centered on the screen. This is a nice React task that doesn't strictly require other libraries, so a good way to improve React fundamentals. 
We also need to decide on a way to control z input, which might be click in the window (just like x and y) or might be connected to the scroll wheel or a UI element. There are disadvantages to each approach, so we should try a few and see what feels most natural. 
Additionally, it would be great to have an in-world cursor for the targeting system, that is a different color from the target once its placed. This will let us see where we're sending the hand to before we actually start sending angles for it. 
Finally, we need to disconnect the click-to-set-target from the click-to-rotate-camera controls. We might be able to bind one or the other to ctrl+click instead, or take any other approach that allows us to control them independently.

Coreyboy1820, 2023-03-23:
@starphys Could you break these tasks into separate issues when you find the time? I want people to know all the different things going on. Thanks!