SJSURoboticsTeam/urc-control-system-2023 #82: Add `sjsu::urc_mission_control` interface

kammce, 2023-03-20:
This will allow applications to eliminate the coupling to specific port or technology for communication. This way the applications do not need to depend on or use specifically the esp8266 or a serial port for getting commands. 

The interface should return the mission control command and NOT return a string that then must be parsed. The implementation handles the lower level details of communicating with mission control. 

There may need to a mission control interface for each sub system, but there may be a possible way to do this with a single interface scheme, maybe using `std::variant`.

Coreyboy1820, 2023-04-20:
@50xp50 This task is the one I was talking about.

Viha123, 2024-01-14:
@Coreyboy1820 has done this

closed 2024-01-14 by Viha123